teachingtechYT/teachingtechYT.github.io #132: temp tower failures

ornery-atmosphere, 2020-09-02:
Not sure whats going on but i can't get the temp tower to print much past the second tier. have tried about 6 times and the stepper motor just stops, I am able to print other temp towers and other models and have no issues at all with them, just this temp tower. Ender 3 pro 1.1.4 silent board, dual gear extruder, otherwise stock.
Fantastic site btw. such a great resource for newbies and the more experienced.

teachingtechYT, 2020-09-02:
Any chance of a pic and posting the gcode to filebin.com ?
The only thing that stands out about your setup in the Creality silent board, which aren't compatible with linear advance. Enabling it can cause the extruder stepper to stop responding mid print. I doubt you've gone this way though?

ornery-atmosphere, 2020-09-02:
Yeah I can post up a pic and put the Gcode there for you later tonight . I don't have linear advance on. Not even sure how to turn it on yet. I didn't realise linear advance it wasn't compatible with the silent board, bummer.

ornery-atmosphere, 2020-09-02:
https://filebin.net/zu8y0os5iyc9o318 there is the gcode.
[image]

Cappytano, 2020-09-02:
Up your extrusion. Just realized this issue. It's in the Gcode generator. If your multiplier is say, .92....increase your extrusion within, say Simplify3d during the print to 108%.

migxmac, 2020-09-18:
I think I'm running into this issue as well.
I uploaded the generated gcode to the online [gcode viewer](gcode.ws) and it estimates I'll use -29 mm of filament!
[image]

The generated gcode(changed type to .txt for upload)
[temperature_tower_200_225.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/5247836/temperature_tower_200_225.txt)
:

teachingtechYT, 2021-07-15:
All of the tests have been tweaked in a coming update. There's nothing to address this because it's a rare problem but more emphasis will be placed on explaining the use of M221 in the custom start gcode to address this.

closed 2021-07-15 by teachingtechYT

ornery-atmosphere, 2021-07-15:
awesome thank, just to let you know the problem disapeared when i changed from the stock creality firmware to my own build of marlin 2.0 bugfix

teachingtechYT, 2021-07-16:
Glad to hear you found a fix. There's so many variations in firmware it's *almost* impossible to keep up.