teachingtechYT/teachingtechYT.github.io #133: Gcode Generator problem temperature and retraction -with potential solution

Cappytano, 2020-09-02:
Having gone through many headaches it finally occurred to me that by the time i get to the Temp or Retraction code, my prints constantly failed. So frustrating.

But then i realized they were under extruding. But after all the config time, it didn't occur to me...because why would it. The reality, i believe is, the Gcode does not compensate for your Slicer's extrusion multiplier.

So  for instance, after fine tuning my multiplier was .87 with a 0.4mm nozzle for simplify3d. It wasn't until i bumped my extrusion multiplier (printing via USB cable and S3D) I bumped my extrusion to ~114 and like magic I was good to go again. Not sure how you can amend this unless you add a form fill box for the gcode generators to account for the individual users' multiplier setting.

teachingtechYT, 2020-09-03:
All of this is explained at the bottom of the flow tab, under the heading: "Can I use this flow value in the other tests on this site?"

At the top of each form it says: 

> If you have additional start commands, tick the box and enter the gcode. This can be used for an extruder prime sequence, overwriting the standard flow rate, setting K factor and more. Tick the box for more details.

and when you click to expand, it says:

> Telling the firmware to alter the flow rate of the gcode to follow. This does not mean the exact flow rate you have set in your own slicer. For example, using M221 S120 would set the flow rate to 120% of what it was originally sliced as in Simpilfy3D. Use this to compensate for obvious over or under extrusion you may encounter with these tests. Additional information available at the base of the Flow Rate tab.

There is a lot on the pages so easy to miss, but is this explanation and the instructions adequate to explain what you experienced?

As for adding the parameter into the physical model of each segment, that would require wizardry way beyond the scope of the site. It would be awesome, however.

On the base layer, many have requested it. Using M221 as above will only affect E values, not Z, and hence shouldn't be a problem.

Cappytano, 2020-09-13:
My mistake! I remember reading that when you first posted this a while back but since had forgotten. Might i suggest a pop-up with a reminder to take this corrective action?

Also - while i have you - I appreciate the videos you post. I've been printing for awhile now, first printer was a Makergear MKII i convinced a previous employer to purchase. Since i've been grounded due to covid, a year ago or so I went with an Ender 3 Pro - quite a step down I must add so I've spent a lot of time trying to make functional improvements and many of your videos along the way have helped! Something to consider in the future - perhaps you can help, i have a lot of issues with Z-axis. I've been over and over the alignment and know my top end is approximately 0.25mm off due to a z-axis support that printed a little thick. Outside of that though, I have a lot of layer shifts in the 200-225mm range though i see hints of it around the 160+. I believe i've for the most part corrected the 160 but the 200+ is still a headache. Perhaps a video going over problem areas could be useful for others. Due note my issues are on taller skinny objects so adding a skirt and additional layer or two to it helps but I still can't solve the 0-2mm shifts that seem to plague that region. Maybe fine tuning (proper torque) the bronzed looking, threaded z-axis piece. Maybe its an issue with the bracket holding the extruder itself - i've heard other people having alignment issues because of it. Maybe this can be added to your webpage of areas to check and include a file for vase mode for a small, tall object like a tube (exactly what i'm having issues with).

Other possible topics for vids or additions to your page if you don't already have them - 

- Proper belt tension (what it should be)
-Pitan/Drakon extuder - i'm going through a Pitan now and its been interesting - a how to vid would be great as documentation seems a little lackluster and I'm sure it will be a worthwhile upgrade.
-Pitan/similar geared extruder with direct drive or direct drive adapters alone --- pro's con's
- upgrading stepper motors? Is it worth it and how to? I have extras laying around and feel the y-axis is under powered for the mass it has to move. Perhaps a larger x-axis stepper is better when going to direct drive from bowden for an ender 3 or similar?

Anyway - that's all for now - thanks for all your doing!

closed 2021-07-13 by teachingtechYT