teachingtechYT/teachingtechYT.github.io #134: Baseline test print (cube) does not contain a skirt

dburr, 2020-09-05:
According to the main page, all generated gcode files should contain

> A single layer skirt (except on the acceleration test)

However the baseline test print (calibration cube) does not.

teachingtechYT, 2020-09-06:
I am unable to reproduce. Here is my gcode preview:
[image]
and this is the actual gcode:

> ; feature skirt
> ; tool H0.200 W0.576
> G1 Z0.400 F1200
> G1 X135.136 Y137.985 F6000
> G1 Z0.200 F1200
> G1 E0 F1800 ; custom un-retraction/prime
> G92 E0.0000
> G1 X137.985 Y135.136 E0.1737 F1800
> G1 X162.015 Y135.136 E1.2095
> G1 X164.864 Y137.985 E1.3832
> G1 X164.864 Y162.015 E2.4190
> G1 X162.015 Y164.864 E2.5927
> G1 X137.985 Y164.864 E3.6285
> G1 X135.136 Y162.015 E3.8021
> G1 X135.136 Y137.985 E4.8379
> G92 E0.0000
> G1 E-5 F1800 ; custom retraction
> G1 Z0.600 F1200
> G1 X135.712 Y138.224 F6000
> G1 Z0.200 F1200
> G1 E0 F1800 ; custom un-retraction/prime
> G92 E0.0000
> G1 X138.224 Y135.712 E0.1531 F1800
> G1 X161.776 Y135.712 E1.1683
> G1 X164.288 Y138.224 E1.3215
> G1 X164.288 Y161.776 E2.3367
> G1 X161.776 Y164.288 E2.4898
> G1 X138.224 Y164.288 E3.5050
> G1 X135.712 Y161.776 E3.6582
> G1 X135.712 Y138.224 E4.6734
> G92 E0.0000
> G1 E-5 F1800 ; custom retraction

Can you check your gcode file for this? If it's there then my guess is there is a severe lack of filament primed in your nozzle.

teachingtechYT, 2020-09-07:
Closed pending a response from the OP.

closed 2020-09-07 by teachingtechYT