teachingtechYT/teachingtechYT.github.io #135: Zhop in the Temperature Tuning test

kromeninja, 2020-09-05:
Hi! Would it be possible to add Zhop as an option in the Temperature Tuning calibration, as we previously calibrate it within the retraction area, I feel like those chosen settings should be able to be then carried over to the next test which is Temperature Tuning.

teachingtechYT, 2020-09-06:
It can be done and I'd like to add it to the subsequent tests. Only trouble is it's not a fun job as currently a lot of manual changes need to take place. I need to write a script to automate this given more nozzle size/layer height combinations are on the way.

teachingtechYT, 2020-09-06:
And by the way the test does have z hop enabled currently, I think 0.4 mm from memory.

teachingtechYT, 2021-07-13:
Z hop is being added to every test in a big update later this week.

closed 2021-07-13 by teachingtechYT