teachingtechYT/teachingtechYT.github.io #136: Is the acceleration testing object printed rotated by 90 degrees?

lmamakos, 2020-09-06:
I might be misinterpreting things, but when I print the acceleration test generated gcode, the side with the "X" character is printed on along the Y axis (bed moving front to back on my Ender 3).  The side with the "Y" characters is printed along the X axis.

[image]

Maybe this is traditionally how this object is printed, first time I've encountered it.  But the faces are not labeled in the same way as, e.g., a typical 20mm calibration cube.

teachingtechYT, 2020-09-06:
This is the intended behavior and I need to add a note explaining.

My logic is that vibration from the y axis is actually shown on the X side. Visualise the Y moving back and forth in your head and hopefully you'll agree.

teachingtechYT, 2021-07-13:
I'm adding a video to the bottom of that page this week to further explain.

closed 2021-07-13 by teachingtechYT