teachingtechYT/teachingtechYT.github.io #137: Inconsistent Result From E-Step Calibration

jormono, 2020-09-06:
Hello,
I could be doing something wrong, following the tutorial/website step by step and I'm at the point of callibrating the E-Step value for my extruder. I ran the test and had about 15 mm remaining so I entered the suggested value from the calculator and did the test again to double check. The second time I had about 25 mm remaining. The Third Time, I had about 5 mm remaining.

I wound up resetting the value to the original which was 471.50

teachingtechYT, 2020-09-06:
It's worth checking that the hobbed gear is free of filament debris and the grub screw holding it to the extruder stepper output shaft is tight.

If those are ok, it may be worth running the test in free air, as in the filament exits the extruder into free air instead of going into the hot end.

teachingtechYT, 2020-09-09:
Closed pending an update from OP.

closed 2020-09-09 by teachingtechYT