teachingtechYT/teachingtechYT.github.io #139: Octoprint plugin adaptation

iFrostizz, 2020-09-07:
Hello TT. First of all, thank you for all your work and for this wonderful 3d printer calibration wizard.

I'm a beginner in programming and enjoy programing Octoprint plugins. So I though that it could be a nice idea to adapt this in an Octoprint plugin, making everything even easier. I already programmed a bed levelling wizard that was especially made for the Ender 3, but it's not really finished because I don't have my 3d printer just next to me and can't ensure myself that it's perfectly working.
What do you think about this ? (this message is also for the community)

Good day !

teachingtechYT, 2020-09-07:
It sounds like a good idea, I've just never authored an Octoprint plugin before. I assume it would embed this site in an Octoprint tab and no actually changes are needed to this site?

I have no problem with that, and give my blessing if someone wants to make it. Happy to add a link to the plugin as well on the front page.

iFrostizz, 2020-09-08:
@teachingtechYT Hey !
Don't need to embed the website, it'll in my opinion not be very handy. Nothing to change on your website excepted, yes, the link if you mind to
Hope I'll succeed :) 
Good day

teachingtechYT, 2021-11-02:
Seems stale, closing.

closed 2021-11-02 by teachingtechYT

iFrostizz, 2021-11-03:
Hi, I haven't seen the tag changes, sorry... But the plugin was already out: https://plugins.octoprint.org/plugins/calibrationcompanion/
It needs some fixes that I will push once I will have a little bit more time