teachingtechYT/teachingtechYT.github.io #145: SUGGESTION: E-step calibration before First Layer and Baseline

wnovinger, 2020-09-13:
Because the E-steps calibration is responsible for controlling the amount of filament extruded as directed by the slicer, this calibration had a direct effect on all of the other flow- and speed-related calibrations. I suggest, therefore that you might want to reorder your calibration steps to take this into account.

Fantastic tool, as we have all said before, Michael! Keep up the good work.

dgentry, 2020-12-09:
I agree.

My Ender 3 Pro, ordered from Amazon for "Black Friday" arrived with a woefully incorrect E-steps calibration (100-ish when it needed to be 190-ish), so I wasted a lot of time trying to get my first layer to work well.  I finally moved on and tried a cube, figured out that it was just a whole bunch of under-extrusion, and moved on to calibrating the E-steps, which made the first layer and baseline cube work really well.

As @wnovinger says, this is a great tool, and I think your systematic framework with community input and fixes will result in it being the definitive filament printer calibration/troubleshooting guide.

Thanks!

teachingtechYT, 2021-07-13:
Agreed. I liked having it next to flow but I can see the merits of this. Changed in update later this week.

closed 2021-07-13 by teachingtechYT