teachingtechYT/teachingtechYT.github.io #146: SUGGESTION: Linear Advance Tuning before flow calibration

Hi-Alex, 2020-09-15:
Linear Advance function is responsible of consistent line width during the print. Because of that we need calibrate it before flow test, so we can be sure that line has same width for each side of cube. This way the value will be correct and there will be no need to increase the flow after this calibration because of the gaps in the filling.

P.S. Thanks for the tool! :)
P.P.S. Sorry for poor English.

Hi-Alex, 2020-09-15:
Oh, forgot to say that LA tab can be marked as "optional" or "if enabled". 

Also, in flow tab need to add text "If you didn't calibrate LA, or it is not enabled in your firmware, use lower speed (20-30mm/s) for following test". 

Every next tab after LA must have option to set LA K-factor. (M900 K0 should be default, I think)

teachingtechYT, 2021-07-13:
I will decline this one due to it confusing less experienced users. Considering on the majority of printers it won't even be enabled without firmware recompilation a new user is going to be completely lost.
I have added a note at the top of the flow tab, however.

teachingtechYT, 2021-07-13:
And by the way, the custom start gcode feature already has instructions for setting linear advance k factor.

closed 2021-07-13 by teachingtechYT