teachingtechYT/teachingtechYT.github.io #148: Slicer Flow Calibration – different thicknesses on walls of cube

drewnoakes, 2020-09-16:
I performed the flow calibration test, printing the one-wall cube and measuring the thickness of each wall. Turns out they're all quite different! 

Any thoughts on what may cause this? Is this something others have noticed. If there's something to be said about this, it might make an interesting addition to the document. It also requires marking the axes before removing from the build plate, so that the orientation is not lost.

Thanks for this great site.

ChrisToxz, 2020-09-19:
What is quite different? Anyway take a average of all sides to do calculation

drewnoakes, 2020-09-19:
Average is easy to do but I think there is an underlying bias between the axes in my machine, perhaps the nozzle or the direction of travel.

From memory the numbers were .42, .48, .51, .55.

ChrisToxz, 2020-09-20:
What machine btw? And did you print with default profile (except for the changes that are requested?, or did you use custom profile?

drewnoakes, 2020-09-20:
CR-10S. Standard profile.

teachingtechYT, 2021-07-13:
I have experienced this also. I don't have an explanation. However, I will add a note to the bottom of the tab.

closed 2021-07-13 by teachingtechYT