teachingtechYT/teachingtechYT.github.io #149: Saving time and resources in retraction test

TheAndi, 2020-09-17:
I was really looking forward to trying the calibration test.

But I was a bit disappointed watching my printer print the "base" for 10 minutes, which has nothing to do with the test at all.

As one will run this test probably multiple times, it would be great to save some time and resources in omitting that base. With good bed adhesion a base would not be necessary at all and if desired (to keep the towers together) a single base layer would be enough. Having a >1 mm base looks nice, but does not help the testing and is a bit overkill IMO. 

Keep up the great work!

sibero80, 2020-11-03:
I was thinking the same thing.
Perhaps I would suggest something like what inket printers do... and that is setting up a range, for all variables, that way, you could have multiple towers, taking advantage of the full bed area, where smaller towers would be printed, but allowing for a wider array of permutations to be tested.

In the end, this could make the test longer in printing time, but would make the test more unnatended, while porably using roughly the same amount of material.

teachingtechYT, 2021-07-13:
An update later this week reduces the base to only 1mm and the overall test id faster too.

closed 2021-07-13 by teachingtechYT