teachingtechYT/teachingtechYT.github.io #150: Can't duplicate retraction success

endobengo, 2020-09-18:
I'm running a mostly calibrated stock Ender 5 Pro(with hotend fix) but can't eliminate stringing and get retraction right.  Using your gcode it comes out near flawlessly at 5.5mm at any speed(tested 30-80mm/s with no noticeable difference) but when I use those exact settings in other slicers like Creality's or Cura it strings like mad.  Uploaded image for reference, 1st-Creality, 2nd-Cura, 3rd your test, all using the same settings.  Obviously I'm missing something, but for the life of me I can't find it.  Any advice?  I'm betting it's something ridiculously simple.  
[image]

closed 2020-09-18 by endobengo

flmgexec, 2021-03-21:
Have you solved it @endobengo? May I ask how?

vicenzajay, 2021-08-16:
I also sent a message on the site about this.  Retraction tunes beautifully using the generated G-codes, but then fails miserably when translated into Cura (exact same settings).  This is maddening....

deendee, 2023-07-16:
I know this is an old post, but has anyone found the answer?  My recently modified to direct drive Ender 3 is good with the TT gcode, but fails with Prusa Slicer and Super Slicer.  I think I have tried everything, but there must be a fix.