teachingtechYT/teachingtechYT.github.io #151: Request to be able to select .6 nozzle.

mecra, 2020-09-18:
It would be nice if the nozzle size can be changed. I've recently changed to .6 and I can't use any of these tests anymore. Thanks!

5teveO, 2020-09-22:
Or even common sizes - such as 0.8 and 1mm or smaller down to 0.2?

wifiguru10, 2020-09-23:
Agreed +1

TwoToneEddy, 2020-10-03:
This would be really handy!!

iFrostizz, 2020-10-05:
I'm currently doing that on my Octoprint plugin, not too hard to change. You can contact me if you wanna beta test :)

vcliment89, 2020-10-14:
Also would like this

bderancourt, 2020-10-27:
Would like this too. Ideally, nozzle size and layer high should be configurable.

Cosantoir, 2020-10-28:
I'm totally +1'ing this I've a volcano hot end id love to dial in and I dont have anything less than a 0.6 and would love to dial that in at least and anything else would be icing on a cake but adding a 0.8 and 1mm would be heaven

johnkraczek, 2020-11-05:
+1 on this request as well.

Glodigit, 2020-11-25:
+1 on this

w0lv3r1nix, 2020-12-11:
+1, add 0.25 0.2 to that 🥇

teachingtechYT, 2021-07-13:
Coming this week.

closed 2021-07-13 by teachingtechYT