teachingtechYT/teachingtechYT.github.io #158: First layer test issue

exostasis, 2020-09-27:
For some reason the gcode generated for the first layer test just spits out tons of filament at the end of the skirt made. Here is the gcode I did add some custom line purging because I thought the issue was at first due to missing line purging since I was making dinner I wasn't actively watching it.
[firstlayer.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/5287681/firstlayer.txt)

vdaghan, 2020-12-30:
I use Anycubic i3 MEGA and I did not have this problem. I am not sure about the generated gcode, so I will just report my take:

Did you notice `G1 XNaN YNaN etc.` lines? I suspect them but could not reproduce them. Tried negative/zero bed dimensions, negative/zero "extra margin from edge", with settings from the file as far as I can:
* hot end temp: 200C
* bed temp: 60C
* ABL/G29
* retraction: 6mm
* retraction speed: 25mm/sec

Nothing. Can this be a browser issue?

JonnyTizz, 2021-04-22:
I just had the same issue, coupled with only one square being generated at the centre of my 220x220 bed. I had the same `NaN` values in my gcode. I refreshed the page and regenerating the file fixed the issue. My two gcode files are attached.

From comparing the settings export txt files I noticed that the "extra margin from edge" property was set to `mm`, with no number, for the broken generation. It was set to `0 mm` in the settings that generated the valid file.

I was able to reproduce the issue (although I did not test on my printer to prove it created the blobs) by clearing the margin field and leaving it empty

### Broken

[image]

[broken-firstlayersettings.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/6361284/broken-firstlayersettings.txt)
[broken-firstlayer.gcode.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/6361279/broken-firstlayer.gcode.txt)

-----

### Fixed

[image]

[fixed-firstlayersettings.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/6361291/fixed-firstlayersettings.txt)
[fixed-firstlayer.gcode.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/6361293/fixed-firstlayer.gcode.txt)

teachingtechYT, 2021-07-13:
There is a default value of zero so I'm not sure how this happened.

closed 2021-07-13 by teachingtechYT