teachingtechYT/teachingtechYT.github.io #159: FEATURE REQUEST: More Temp Tower Options

qwksilver, 2020-09-27:
1: preset temps for petg, pla, abs, and tpu, 
2: custom target (ie 240pla, would go +-5 & +-10 on a 5 layer target, automatically
3: tower invert (so high to low or low to high (for use with preset/range buttons)
4: more tower layers, eg 5(default), 7 or 9 if needed, when used with preset/range would add +-15 for 7layer and +-15&20 for 9 layer

i'm having a hard time tuning in petg on a moded ender 3, and this code generator would be nice, for today i will likely just find a large tower and setup manually.  but i will run this one first.

johnruciak, 2021-01-07:
Main one for me would be print speed on the temp tower page. Having issues dialling in ApolloX (mainly cosmetic). Great resources all round, BTW TT

teachingtechYT, 2021-07-13:
You have to type in six numbers, that's it. The time involved in writing a GUI to insert these for you, and then the consequent debates from people over what the default suggested temp should be for a given filament are just not worth it.
John, use the controls of the printer to raise or lower the feedrate as you see fit.

closed 2021-07-13 by teachingtechYT

johnruciak, 2021-07-13:
no worries!