teachingtechYT/teachingtechYT.github.io #165: Retraction tower GCODE attempts horizontal movement immediately after delta homing

hitjim, 2020-10-08:
https://teachingtechyt.github.io/calibration.html#retraction

On my delta printer, the GCODE never got farther than homing, then attempting a horizontal movement before out of the "safe zone"

I'm running Klipper, not Marlin, but so far the GCODE has been working. I believe Marlin automatically does a Z move to lower all effectors after a home, but not all firmwares will do that.

Thankfully Klipper simply errors on an invalid move.

[retractionsettings.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/5344598/retractionsettings.txt)

zolkin, 2020-10-17:
I ended up swapping XY movement GCODE line with the next Z movement one, which fixes the issue for me.

hitjim, 2020-10-20:
Yeah, a good workaround, and I ended up doing that as well. If it's within the control of the GCODE generator, seems like a safe change to make for all motion types.

MeanEYE, 2020-11-03:
Wish I had seen this comment prior to nozzle slamming against print bed. So to confirm solution is to swap first G1 X# Y# with line that follows which is G1 Z0.2. Z axis needs to be set first before moving XY.

Firworksyt, 2020-11-22:
Also just had a bed slam on my delta and the hot end burned a hole through my Buildtak sheet by the time I ran over to it and shut it off. Rather than just swapping the Z and the XY lines, the temperature tower moves down to a slightly higher height above the bed then does the sideways move alongside moving down to the final height.

`G1 Z0.400 F1200
G1 X-25.695 Y-12.055 F6000
G1 Z0.200 F1200`

Might be a good option as well since that's already what another one of the tests is doing.

teachingtechYT, 2021-07-13:
I have a fix for this in an upcoming update. To accommodate deltas homing at Max, before anything other movement, there is a hard coded G1 Z1 to lower the nozzle above the bed.

closed 2021-07-13 by teachingtechYT