teachingtechYT/teachingtechYT.github.io #166: Need either more brim or primer line

brianhole47201, 2020-10-08:
All of the g-code really needs either a bigger brim or a primer strip.  By the time my nozzle heats up and drips out what's ready to go it takes every bit of the brim just to get filament flowing for the actual print.  Most of the prints don't use anywhere near the whole bed, so if you just want to avoid bed clips move the prime strips in much further.  Or give the option to print a normal prime stripe for those who don't have clip issues.

asusdk, 2020-10-09:
You can add additional startcode, ive done that with a big primer line =)

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

teachingtechYT, 2021-07-13:
This was in the custom start gcode instructions all along. 
You might think there is a universal prime gcode sequence but there is not. Bed clips, and more importantly delta printers won't be compatible.
As the instructions say, copy your sequence from your slicer and it will stay between tabs.

closed 2021-07-13 by teachingtechYT