teachingtechYT/teachingtechYT.github.io #167: Flow difference ?

asusdk, 2020-10-09:
Hi, and firstm, thanks a lot for your effort to build this calibrationsite !

I have the problem, that just with the codes from this site, all my (first)layers seems way to thin. Do this site use, a lower flowrate than 100% ?  Every time a run a test with an Code from Cura, the First layer looks damn near perfect, but with the codes from here, the layer is way to thin (tiny tiny little gaps between lines)

Kind regards and thanks in adavance

legend069, 2020-10-30:
mine is the same, but from here is bigger gaps.
sliced in cura is perfect

creativeDev6, 2021-01-04:
@asusdk @legend069 As stated on https://teachingtechyt.github.io/calibration.html#flow _Can I use this flow value in the other tests on this site?_ 
>Let's say your old flow rate was 100% and you have tested and corrected this to 96%. **The gcode on this site originally had a flow rate of 90%** when sliced, so applying your 96% to that gives a final result of 86.4%, not 96%. Your slicer profile settings will also be different in other ways, which further complicates matters. Therefore, there is not a straightforward correlation between your slicer and my gcode generators.

By this I assume the site uses a flow rate of 90%. You can adjust it by the following formula:

`slicerFlow / 90 * 100  = adjustedFlow`

And use `adjustedFlow` with the `M221` in the additional start gcode box:

`M221 S[adjustedFlow]`

So assuming you have 100 flow in your slicer you would write the following (but usually 100 flow is too much but this will also depend on your extruder steps):

[image]

@teachingtechYT I would suggest to build this formula into the site to also adjust the flow dynamically.

teachingtechYT, 2021-07-13:
Thank you creativeDev6 for explaining.

closed 2021-07-13 by teachingtechYT