teachingtechYT/teachingtechYT.github.io #175: E steps callibration

sbaeder, 2020-10-23:
Doesn't know how to handle LARGE or decimal input for steps.   I have a geared stepper and using micro stepping, I need a number like 470 to get 100mm extruded.

But overall, good site!

sbaeder, 2020-10-23:
Looked at this more, and I think all we need to do is mover the multiplication to before the divide and add ()

OLD....
`function esteps(){
    var oldSteps = document.estepsForm.oldSteps.value;
    var remainingFil = document.estepsForm.remainingFil.value;
    var actualExtrusion = 120 - remainingFil;
    **var newSteps = (oldSteps/actualExtrusion*100).toFixed(2);**
    $("#e1").html(remainingFil);`
New...
`function esteps(){
    var oldSteps = document.estepsForm.oldSteps.value;
    var remainingFil = document.estepsForm.remainingFil.value;
    var actualExtrusion = 120 - remainingFil;
    **var newSteps = ((oldSteps*100)/actualExtrusion).toFixed(2);**
    $("#e1").html(remainingFil);
`

teachingtechYT, 2021-07-13:
I'm sorry but I'm not sure what the problem is.
The calculator works with large or small e step values.
The proposed code change doesn't actually change the output due to mathematical order of operations.
Maybe you can reopen with information such as what you input, what you expect to receive and what you actually receive?

closed 2021-07-13 by teachingtechYT