teachingtechYT/teachingtechYT.github.io #182: RETRACTION - Results of test not translating to real life printing.

sibero80, 2020-11-03:
I managed to get a settings narrowed down to a pretty stable result using the RETRACTION TAB.
I'm usingn an Ender 3Pro with Direct Drive Extruder, and this were my resulting settings.
Settings for retraction tuning form
```
_________________________________________________________________________

All changes are marked in the gcode with 'custom' at the end of each line. Open the gcode in a text editor and search for this to your inputs if needed.

Bed: 200 x 200 mm

Temperatures:
Bed: 60 deg C
Hot end: 193 deg C

Part Cooling: 100% fan from layer 2

ABL: Probe new mesh at the start of print - G29 (BLtouch,EZABL,etc)

Segment | Retraction distance | Retraction speed | Extra restart distance | Unretract speed | Z hop
   F    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
   E    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
   D    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
   C    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
   B    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
   A    |          1.2 mm       |     30 mm/sec    |        0       mm      | 5 mm/sec       |  0 mm
```
[image]

Here you can see the progression on how stringing got better each iteration.
Then, I went to print this model [Random Retraction Test](https://www.thingiverse.com/thing:607968) translating the settings from the RETRACTION tab.
My results were very underwhelming.
[image]
On the left is the one using the results of the RETRACTION tab, and on the right is the results using the DEFAULT cura profile.

I'm I doing something wrong?

RNewDeal, 2020-11-06:
Had the same issue, tower was 0.2 to 1.2 and had perfect retractions all the way, and in real printing I'm up to testing 2.0 retraction. A description of all the settings for these tests would go a long way.

AalianKhan, 2020-11-13:
I'm having the same issue with my cr10 v2 direct drive. However when I put in the same STL and retraction setting in prusa slicer I get a perfect result which correlates with the result I got with the test.
It also seems that this is only happening with direct drive but that may be wrong
@RNewDeal are you getting good results with 2.0mm retraction

RNewDeal, 2020-11-13:
@AalianKhan I still had garbage results with 2.0 retraction and this is a direct drive machine. What fixed it for me is a single factor, which was zhop speed. Cura doesn't allow you to set it above 10, and Ideamaker doesn't allow you to touch it at all, but S3D standard is at 20mm/s. Once I used S3D I found stringing disappeared despite identical settings, confirming the zhop speed being a key factor.

B0IIE, 2020-11-17:
@RNewDeal I´m having the same issue as you two. But I can´t find a setting called zhop in S3D. Could you post a screenshot?

dennisracine, 2020-12-09:
Likewise, my test prints from the generated gcode look awesome, but I can't get S3D slice of the same file to replicate the test. Very fustrating.

dennisracine, 2020-12-09:
[image]
Left test is from the generated gcode, right is my S3D slice with the same? settings...

lopez453, 2021-02-22:
I've had same issue today. Generated code printed beatifully on my direct drive (geekdrive) modded ender 3pro. 
tested PETG at 240C with 0.6, 0.8, 1.0, 1.4, 1.6 and 40 speed all without stringing.
so 0.6 was fine for me (most smooth seam), why go higher.
Sliced in cura - bad result. Compared with gcode analysers, turning off coasting and making seam for center of object make the gcode trajectories and speeds almost same but there is one significant difference:

retraction from **generated gcode** with 0 extra retract distance:
retract specified length of filament, lift Z, move to target, lower Z,  **deretract 0!!!!!!**, continue printing

retraction from **cura gcode**(with 0 mm3 retraction extra prime amount):
retract specified length of filament, lift Z, move to target, lower Z,  **deretract what was retracted!!!**, continue printing

I've just replaced all CURA deretractions with 0 and the reprint is ongoing, will post results.

RNewDeal, 2021-02-22:
@teachingtechYT I think the solution for this would be to drop the z hop speed to whatever Cura uses stock, which is like 10-5mm/s

girrisgod, 2021-04-11:
using microswiss dd set up and like many have said when i translate my very good results to prusaslicer i get crazy stringing so what gives. what is s3d doing that prusa wont

teachingtechYT, 2021-07-13:
A couple of things:
- There is a run down of the base settings from the profile on the introduction tab. I am soon to attach the simplify3d fff profile so users can see any setting they want.
- I assume you have all worked through the list on the retraction page under "Other factors beyond the scope of this test - Important!". I am soon to add some new parts to this list, most significantly to check for auto cooling or slow down. I will add an item about Z hop speed too based on this discussion.

closed 2021-07-13 by teachingtechYT