teachingtechYT/teachingtechYT.github.io #183: Retraction:  Goals unclear

DonutATX, 2020-11-06:
Doing the retraction test on my BIQU-B1, after doing the stuff before.  

I didn't change anything for any of the levels.  I get no threads, just a little fuzziness.  

Reading the page over and over, I realized that I can't determine what the goal of this test is - what do I want to see?  No threads?  Thin threads?  No fuzziness?

A section on "Goals" or "Desired Results" and a picture of "Perfect Calibration" would be great.

Thanks!

TownDrunk, 2020-12-25:
On PLA, I found my edges became smoother on the close edges.

teachingtechYT, 2021-07-13:
If you are using the site, I think it would be self evident that you are trying to improve your print quality. For the retraction page, that means minimising stringing. An example of stringing is given near the top of the page, so it should be clear what that is.
Have you read the "Interpreting results" section at the bottom of the page? It shows a clear example of a completed test with desirable and undesirable segments, with a suggested follow up course of action.

closed 2021-07-13 by teachingtechYT