teachingtechYT/teachingtechYT.github.io #185: Problem with GCODE

trikeus, 2020-11-07:
Newbie here, with a BL touch installed on my printer. Whenever I download the GCODE and run it on my printer, it goes through the bed-leveling mesh probing, but after touching down on the 9th mesh position, the printer goes back to zero x axis as if it is done printing.

I've tried this on multiple different tests, but all are the same: probe for auto-level, then end print. What do I have wrong here?

P.S. Michael, I cannot describe how appreciative I am for all of the work you have done for the community. Wouldn't be here without you!

teachingtechYT, 2021-07-13:
Hi there,
Perhaps you could post the gcode that the site made for you on pastebin or similar?
Also, have you imported it into your slicer to view the preview? It is possible that an erroneous value in one of the forms is breaking the generator and the file is largely empty.

teachingtechYT, 2021-08-23:
Stale

closed 2021-08-23 by teachingtechYT