teachingtechYT/teachingtechYT.github.io #187: Ordering of processes

stdekart5827, 2020-11-07:
Just some food for thought: This is with the assumption the processes should be followed from Top left to right, then onto the second row.
Would it be worth having the e-step calibration as the first step? Being that this could effect the first layer and baseline print pending how far it's off? Otherwise you may end up having to go back to the first layer after the e-step calibration due to the change in the amount of filament extruded.

SoongJr, 2021-01-12:
Yes, I came here to ask this as well. I just spent a hell of a lot of time on manual bed levelling, and now got to flow rate section and noticed that my finely tuned extrusion was all for naught because we're now changing the extrusion rate 😂

SoongJr, 2021-01-12:
Oh, as the note below the Flow section points out however, the flow is of course set in our slicer and doesn't affect the generated gcode... So just putting the Flow section first will not change anything for the bed levelling section... For that to be possible, the gcode generator would need to apply the flow rate multiplier to the source gcode. I don't know how straightforward this is...

But at least a note in the levelling section that you don't need to get it perfect would be good 😂 At least not all work is lost, I can still apply an offset to the mesh values once the flow is correct and I've sliced my own bed levelling test...

teachingtechYT, 2021-07-13:
This has been moved.

closed 2021-07-13 by teachingtechYT