teachingtechYT/teachingtechYT.github.io #188: [Feature request] Pursuing efficiency: calibrating XYZ-steps with the "baseline" cube

anesuccio, 2020-11-09:
My suggestion aims at **improving efficiency and print time**: the now called _Baseline Print_ could be used to calibrate X Y Z steps for dimensional accuracy.

I usually do this type of test with a single wall model, so the output G-Code would be similar to _Slicer Flow Calibration_, but I think the same calibration could be achieved with the _Baseline Print_ cube, after flow and E-steps are set correctly.
The correction formulas are the same for E-steps calibration.

End note: I would like to thank Michael for this useful website and motivation he gives me in every video.

Dublin1, 2020-11-17:
I agree on the above. As I'm quite new to 3D printing some advice would be appreciated to what method of correcting XYZ-steps that is preferred (in the printer or in the slicer)?

vguttmann, 2020-12-05:
I don't think that this would be very accurate, at least on bowden-style printers, thanks to remaining pressure. If you take a look at the "good results" photos that Michael made, on one you can see the blobby corners, which would throw off your measurements significantly, so this should only be done after configuring linear advance. This would then defeat the purpose of using the baseline print at all.

teachingtechYT, 2021-07-13:
This didn't exist when this issue was first posted but an XYZ calibration tab has now been added.

closed 2021-07-13 by teachingtechYT