teachingtechYT/teachingtechYT.github.io #191: Temperature tower does not wait until nozzle temperature is actually set to desired temperature

DavoudTaghawiNejad, 2020-11-15:
The gcode users m104 instead of m109 to change the temperature, this does not guarantee that the temperature is the desired temperature for this layer and invalidates the test, somewhat

vdaghan, 2020-12-30:
gcode I was provided with goes like:
> M140 S105 ; custom bed temp
> M190 S105 ; custom bed temp
> M104 S200 T0 ; custom hot end temp - first layer
> M109 S200 T0 ; custom hot end temp - first layer

Given last update was on September but this issue was opened in November, I am quite perplexed.

SoongJr, 2021-01-12:
Yes @vdaghan that's correct for the first layer, but later sections look like this:
> ; process Color1-6-3
; layer 91, Z = 18.200
**M104 S190 T0 ; custom hot end temp - C**
; feature inner perimeter
; tool H0.200 W0.480
So it just sets the new target temp and continues printing.

That said, raising the temperature by 5°C isn't a huge task and is probably done by the time the first layer is done printing, so unless you're testing much larger steps the difference is negligible. And you could also get in trouble with burning your print if you just keep the nozzle there for 10 seconds until the extruder is up to temp, unless you park the nozzle somewhere, which would then ooze out all the filament and you have to purge before continuing the print... My guess is this is the main reason not to use M109 here.

DavoudTaghawiNejad, 2021-01-13:
If you want to calibrate your printer, you should worke precisely,
especially if it comes at no cost.

teachingtechYT, 2021-07-13:
If I use M109 instead of M104 to ask the printer to wait for the new temp, it means it will be sitting there oozing until it reaches this temp. The resulting blob will be worse than the current situation and the nozzle might collide with it and knock the print from the bed.
SoongJr is spot on. Unless a park and purge sequence is added, which will of course need to suit all types of printers (including deltas) using M104 is superior. For most printers the new temp will be reached comfortably before that layer is done. So at worse you will have the lower 0.2mm of that 5mm segment not be completely representative of the temperature. That is an insignificant compromise compared to the alternative.

closed 2021-07-13 by teachingtechYT