teachingtechYT/teachingtechYT.github.io #194: provide the configuration of slicer used to generate the baseline

moustic999, 2020-11-18:
I have an almost perfect XYZ calibration print when I use the G-code provided on the site. However, if I take the stl and slice it myself I have some ugly artefact. I'm trying to find the issue, to adapt settings in s3d, but I always have same artefact. 

providing the profile used in s3d would be a good basis to start our own config.

topgun0dk, 2020-12-04:
Yes please...
When i put the same values into Simplify 3D, i do not get a result as good as the gcode from this tool.
Can you share what settings you have put into the Extruder and Advanced sections ?

trippd6, 2020-12-05:
Same here... please provide profile from simplify 3d... I get great results on the retraction calibration test... then I attempt to slice it myself in simplify 3d with the same settings as mentioned on the first tab and get bad results.

Orion2507, 2020-12-15:
Same thing. I went through all the calibrations, but once I move back over to my slicer profile in Simplify3D after inputting the values I no longer produce flawless prints. 

I have tried to replicate the G-code from the output file on the web site but I have not been able to repliacte the quiality the pre-generated slicer creates. The pre-generated G-code does not include the process settings at the beginning. A copy of the Simplify 3D FFF profile would be greatly apprciated.

teachingtechYT, 2021-07-13:
Hi guys, big update under way and I will add the baseline profile to the site when it is published.

closed 2021-07-13 by teachingtechYT

De-Ma, 2021-11-16:
Hi, i dont think this is fully resolved yet?!

I also still have problems transfereing the results.
I Printed some Retraction towers downloaded from the site until i was satisfied.
Tried to translate every setting (yes including start and stop g-codes, and all overall settings from the intro page, basically all settings i could find on the site to S3D) and my results are awefull.
The downloaded File prints perfektly PETG-test towers.
Transfered Settings result in garbage prints (planty of stringing).

I also downloaded your FFF-Profile with no luck:
* The profile does not include the starting and stopping G-codes (i manualy added those)
* Fan speed is set to 1 (1%) instead to 100 (100%) which is noted on the site
* The profile doesen´t un-retract, so it litterly retracts the fillament out of my bowden extruder and does not print the towers because with 3mm retract no more material comes out the nozzel^^ (im not eaven shure how it does that, because i did not find any settings that would do something like that)
* The profile is basically unprintable (would realy be nice if it would be ready to export some useful g-code)

I realy love the g-code feature on the website, but without the possibility to transfer it to my Slicer (S3D) its sadly not that useful.

Maybe its possible to also include the S3D settings in the Downloaded G-code (like in normal exported G-code from S3D) ?
this would realy help figuring out whats going wrong. 
One could also then just import the G-code file into a FFF-profile, at least on S3D and the nwould get the exact same results.

Ore maybe i dont see what went wrong, would realy appreciate the help.