teachingtechYT/teachingtechYT.github.io #195: Possibility to change the nozzel size

zsolt320i, 2020-11-21:
Hi!

First of all your webpage is very useful, thanks!
Unfortunately i cannot use because as i see all the test is configured for 0.4mm nozzle, and im using 0.5 mm.
Would it be possible to insert a feature with nozzle diameter selection, or insertion?

Thx!
Zsolt

zsolt320i, 2020-11-28:
Hello everybody. Any opinion, any statement, what do you think?
Everybody of you are using 0,4mm?

Mrdindon, 2020-12-10:
I agree with you, nozzle size change would be really nice 👍

xchg-dot-ca, 2020-12-16:
Nozzle size is very important.
I have 2 printers with 0.5 and 1mm.
Believe you or not it was easier to have second printer for faster prints then fiddle with nozzle changes and recalibrations every time I need faster print time.

I do vote for different nozzle size across all calibration tests.

Vivida1, 2021-01-02:
Noticed this today as well. Was about to use the scripts to generate the gcode when I realized that there are a lot of options but no option to change the nozzle size...using 0.6 myself.

Ansari-Tech, 2021-01-05:
Agreed with everyone here, I think it would be helpful. I use a 0.3mm nozzle and I notice I get underextrusion(?) using the generated gcode from the site.

Yury-MonZon, 2021-02-07:
+1 for this, as I mainly use 0.8  nozzles. Thank you!

Any updates on this?

Sjoerd, 2021-02-12:
I'm using an 0.6 nozzle on my printer, would be nice to see an update for larger nozzles :)

PatBriPerso, 2021-02-22:
I just change my 0.4 nozzle for a 0.8 one to faster prints and want to calibrate using this wonderful site when I realize that the whole site works only for 0.4 nozzle. Would be great if we can set the nozzle size.

Ansari-Tech, 2021-03-14:
@teachingtechYT can you confirm that reslices of all the test models would be needed to implement this feature? Given a set of slices, one for each common nozzle size, it should be trivial for myself or someone else to implement this into the GUI.

teachingtechYT, 2021-07-13:
Duplicate request but highly requested. Coming in update at the end of the week.

closed 2021-07-13 by teachingtechYT