teachingtechYT/teachingtechYT.github.io #196: retraction on delta delta

Sardau, 2020-11-22:
Hello, the move after the homing in the retraction.js is a XY movement. that cause problem on deltas

G1 X18.872 Y37.041 F6000
(==> become  G1 X-31.128 Y-12.959000000000003 F6000 after  the page procession for delta but the problem persist )

it is beyond the move limit at homing position => the delta motors lose steps and the print fail.

wings2004, 2020-12-12:
I'm getting the same issue on my FLSun Q5 with a Duet3d board. Specific error is - "Error: G0/G1: target position not reachable from current position"

rlatimer4, 2021-01-19:
My printer also (FLSUN QQ-s) has the same issue, only I'm using Klipper.  Klipper fails and states: "move is out of range"

Simply changing these two lines in the gcode so that Z moves first like below fixes the issue:

G1 Z0.200 F1200
G1 X-31.128 Y-12.959000000000003 F6000

teachingtechYT, 2021-07-13:
A workaround for this in included a in a major update at the end of this week.

closed 2021-07-13 by teachingtechYT