teachingtechYT/teachingtechYT.github.io #206: Baseline Cube is in "Wrong" Orientation

dgentry, 2020-12-09:
This might not matter for the way you use the calibration cube (since you don't suggest that people calibrate X and Y steps), but it did cause me to chase X/Y calibration needlessly.

I had previously calibrated my steps with a cube that I sliced myself, and I oriented it on the bed so I could measure the face labelled "X" to adjust the X steps, and the face labelled "Y" to adjust the Y steps.

The cube you generate needs to be rotated 90 degrees left or right so the "X" lines up with the X axis and the "Y" lines up with the Y axis.

Thanks!

combolek, 2021-01-03:
I think this is intentional so measuring all axes is consistent. To calibrate X you measure in the direction that you are looking at X and same for Y and Z. If it was oriented the way you suggest then measuring Z would be different than X and Y (obviously you don't calibrate Z by measuring the face labelled "Z").

teachingtechYT, 2021-07-13:
The cube is placed a the original creator intended. The X face parallel with the x axis and the same for the Y. As you say, nothing on the page suggests using this print for measuring x and y steps.
Since this issue was raised, I have added a specific tab for that.

closed 2021-07-13 by teachingtechYT

dgentry, 2021-07-13:
That’s a good solution, thanks!

mdavis-xyz, 2023-03-03:
> I think this is intentional so measuring all axes is consistent. 

It's not consistent though. At least not for an Ender 3 viewing the gcode in Cura.

> To calibrate X you measure in the direction that you are looking at X

Assuming you mean measuring from the closest side to the furthest side, no that tells you the Y dimension.

If you want to measure the z axis, you put one side of the calipers on the letter Z, and the other side of the calipers on the opposite face.

If you want to measure the x axis, you do *not* put one side of the calipers o the X and the other side on the opposite face. Doing so is a measure of Y. (With the current orientation.)

i.e. it's not consistent.

> The cube is placed a the original creator intended.

Yes, it is as the original creator intent. There's lots of comments on Thingiverse saying the original STL is in the wrong orientation. *If* the original is in a confusing orientation, what's the harm in improving that on this site? (Assuming that implementing the rotation is technically straightforward.)

Prior to coming across this website, I was downloading that STL and fixing the rotation myself in Cura when slicing. If I stick to that approach, what would I be losing, compared to using this page's gcode generator?