teachingtechYT/teachingtechYT.github.io #226: XYZ Dimensional Accuracy Calibration

PantherIT, 2021-01-06:
Hi Michael, this is a great resource thank you so much for putting this together. I noticed you don't have a section on calibrating XYZ dimensional accuracy and was curious why you left that section out? Is there a specific reason? If not I would be more than happy to contribute by creating that section but wanted to check with you first in case there is some reason I did not think of.

DaBayer, 2021-02-13:
Hi, Michael! I second all above - great and HUGELY helpful resource although I feel it needs a dimensional accuracy calibration. It's a step I will insert into your published process when I execute it FINALLY setting up my Ender 3 Pro. I may find that it's not necessary, but definitely worth the check.

dwhempy, 2021-03-31:
Good Evening Michael: I'd like to be the third to endorse an XYZ calibration section. It seems to me that this should be a part of the baseline print area. I haven't been through everything on this site BUT I did watch the video you did on this resource and I plan to go through many of the steps (after I get my XYZ calibrated that is!). Thanks and Cheers!

teachingtechYT, 2021-07-13:
Hi guys, only just reading this now but hopefully you've seen I added this in March.

closed 2021-07-13 by teachingtechYT