teachingtechYT/teachingtechYT.github.io #237: Please add how to calibrate dual Z screws

theschles, 2021-01-19:
Hi there,

Thank you for such a great tutorial of the Ender 3. I have a Sovol SV01 with dual Z screws connected to separate stepper motors -- but driven by a single stepper driver on the mainboard -- so Marlin's G34 command is not an option unless I change system boards (I didn't realize this when I bought the Sovol's "silent board" add-on).  The closest I've gotten is [this video from myfordboy](https://youtu.be/ilezW1FQ_JU?t=163) where he does some measurements and adjustments to align everything.

Would you please consider adding a section and video specifically for calibrating dual Z screws -- both:

1. with a single driver, and
2. with multiple drivers (with which I'm assuming one can do the G34 command)

(PS I added this originally as a comment on the assembly video, then found the Github repo...so since this is probably a better place for this request, I deleted that video comment.  I hope that's ok)

teachingtechYT, 2021-07-13:
Only reading this now but hopefully you saw this one when it was released: https://youtu.be/6UgT9YqY3UA

closed 2021-07-13 by teachingtechYT

Homer-S, 2023-01-11:
Hi!

I had a question about that video, specifically G34. Assuming you are on a printer without a fixed bed, doesn't G34 for calibrating a dual z-axis with dual drivers align the gantry to a moving bed that itself is trammed using the z-axis? I would expect for the bed tram to be accurate the z-axis gantry would have to be trammed to something fixed, either the upper frame or the lower frame of the printer.

Homer