teachingtechYT/teachingtechYT.github.io #240: Core XY

Spacefan-jrb, 2021-01-22:
The Z axis seems wrong for Core XY machine where the bed needs to travel down instead of the gantry traveling up.

Nuttavoot, 2021-02-15:
> The Z axis seems wrong for Core XY machine where the bed needs to travel down instead of the gantry traveling up.

I am using a Core XY machine and I don't see the probem with Z axis. In which test is the wrong.

Spacefan-jrb, 2021-02-18:
Thank you for your looking into this for me.

 

The retraction tuning test.  I have a Project R3D Daedalus printer and thought I would see if retraction needed to be tuned to help with minor stringing and getting supports that did not fuse with the print so bad.  I added the start-up code from PrusaSlicer to the beginning  and submitted the generated code to the printer, with each layer the extruder seemed like it was bringing the bed up and not down.

 

From PrusaSlicer 2.1.0

 

Start G-Code

; Multi-temp start code - 5 September 2020

; Raise bed temp to profile requested bed temp

M140 S[first_layer_bed_temperature] ; Raise bed temp to profile requested bed temp

M104 S150 T0 ; Raise hot-end temp to 150C for G32 auto-calibration

M190 S[first_layer_bed_temperature] ; Wait for bed temp to rise

M109 S150 T0 ; Wait for hot-end temp to rise to 150C

G28 ; home all axes

G32 ; Autocalibrate bed

G32 ; Autocalibrate bed a second time

G28 Z ; Home Z after bed leveling

G29 S1

G1 Z20 F600 ; Move bed down 20

M104 S[first_layer_temperature] T0 ; Raise hot-end temp to profile requested hot-end layer 1 temp

M109 S[first_layer_temperature] T0 ; Wait for hot-end temp to rise to profile requested hot-end layer 1 temp

;

G1 X0.0 Y0.0 Z1.0 F2000.0 ; prepare to prime

G92 E0 ; reset extrusion distance

G1 Z0.2 F600

G1 X60.0 E9.0  F1000.0 ; priming

G1 X125.0 E12.5  F1000.0 ; priming

G1 Z0.5 F600

G92 E0 ; reset extrusion distance

; end of start code

 

 

End G-Code

 

; End G-code

M400                                    ; Wait for all previous moves to complete

M104 S0                               ; turn off extruder heater

M140 S0                               ; turn off bed heater

M106 S0                               ; turn off parts cooling fan

G91                                       ; Relative move

G1 E-1.0 Z+20.0 F1000.0     ; E : Nozzle Up/Bed Down 10mm and retract - (E to exclude from layer count)

G90                                       ; Absolute moves

G1 X10 Y280 F2000             ; Move Y-carriage for part removal

G4 S30                                  ; Wait 30 seconds (since M84 seems ignored with arguments)

 

 

 

Before layer change G-Code

 

;BEFORE_LAYER_CHANGE

;[layer_z]

 

 

After layer Change G-Code

;AFTER_LAYER_CHANGE

;[layer_z]

 

Again, thank you for any information you can provide.

 

I am using a Core XY machine and I don't see the problem with Z axis. In which test is the wrong.

teachingtechYT, 2021-07-13:
The tests on this site simply give Z values in the gcode. Whether than moves the bed up or down in absolute terms depends on the firmware. Hopefully you have found the problem is the code you posted above.