teachingtechYT/teachingtechYT.github.io #241: Too many decimal places

Mooselake, 2021-01-24:
I've only made it through the firstlayer and baseline tests, but am finding many values with far more decimal places (could be getting close to atomic diameters) in both generated files than are needed.  For example

G1 X45.364000000000004 Y43.686 E2.5643
G1 X43.686 Y45.364000000000004 E2.6666

You might want to be rounding those off to 4 or 5 decimal places

ryamoo, 2021-01-25:
I'm hitting an issue with this - printing through Octoprint, and the nozzle just stops doing the first wall of the retraction test, just after doing the brim. 
Not entirely sure if it's this line:
```
Send: N361 G1 X125.90899999999999 Y148.122 E6.2761*96
Recv: ok
```
and the "ok" is it just being buffered, or this line:
```
Send: N366 G1 X126.72200000000001 Y147.06799999999998 E0.0187 F1350*54
Recv: Error:checksum mismatch, Last Line: 365
Recv: Resend: 366
Recv: ok
```
which gets repeated until Octoprint disconnects.

teachingtechYT, 2021-07-13:
It's a quirk of the javascript manipulation but this is the first time anyone has said it was a problem.
If you manually trim the extra zeros for several of the lines in a row and run the gcode, it would confirm that this needs attention.

ryamoo, 2021-07-13:
The majority of the generated gcode needed editing because there were lines with too many decimal places throughout the whole file. 
It also occurs on more than 1 of the tests that generate gcode. 

I went the route of downloading and editing a local copy of the site, and changing the code to round the calculations to fewer decimal places. 
I wasn't familiar enough with GitHub to submit the change though, and I wasn't sure I covered all instances.

teachingtechYT, 2021-07-13:
I'm about to do a major update so any pull request will shortly be redundant. Please hold on for a week or so and then we can re-examine.

teachingtechYT, 2021-07-19:
X and Y coordinates now are rounded to fur digits.
Should be plenty accurate and produce cleaner output.

closed 2021-07-19 by teachingtechYT