teachingtechYT/teachingtechYT.github.io #244: Bug in retraction tuning when setting all retraction distances to the same value

ArrMiHardies, 2021-01-26:
There is a bug in the retraction tuning when setting all retraction distances to the same value (so you can testother values after finding a good distance). The final gcode appears as though the test will print in the middle of the bed, but when you actually start the print, it will print on the far back left of the bed (at least on my machine) and in my case slightly off the bed to the left and close enough to cause some minor damage to it.
Change any of the values in the retraction distance at all and it prints in the middle again. This leads me to think there may be a divide by 0 error ini relation to the retraction settings that is somehow getting applied to the print location or bed size that S3D isn't noticing.

teachingtechYT, 2021-07-13:
Can you please post the gcode generated by the site on pastebin and link here?
That will let me import into the slicer to preview (have you done this?)
By the way I've used the form in the way you describe without problem:
[image]
[image]

ArrMiHardies, 2021-07-14:
When I said that it appears that it will print correctly in the center of
the bed, I meant that is where it appears in the slicer, so yes, I have
loaded it into a slicer (Simplify3D) but the location the print actually
starts is different than the preview and is on the back left corner of the
bed and digs grooves into the print sheet. This only happens if all the
retraction distances are the same. If even a single one is different, the
tower prints normally and this doesn't happen.

teachingtechYT, 2021-07-15:
If the slicer preview is correct that should indicate that something is awry with the printer. I've had such issues before after accidentally leaving relative movement mode on (G91).
Hopefully the update being released in a couple of days fixes this for you. Please reopen with pics if it doesn't.

closed 2021-07-15 by teachingtechYT