teachingtechYT/teachingtechYT.github.io #249: Enable Cold Extrusion

gary325, 2021-01-30:
In this section: Extruder E-steps Calibration
https://teachingtechyt.github.io/calibration.html#esteps
Mention that you can enable cold extrusion to send filament without heating up the hot end.
Please verify the gCode: (from Marlinfw.org)
M302         ; report current cold extrusion state
M302 P0      ; enable cold extrusion checking
M302 P1      ; disable cold extrusion checking
M302 S0      ; always allow extrusion (disable checking)
M302 S170    ; only allow extrusion above 170
M302 S170 P1 ; set min extrude temp to 170 but leave disabled

fsgl1tch, 2021-07-10:
I had the same issue initially that the printer reported: `echo: cold extrusion prevented`. Sending the command M302 P1 disables the check. **NOTE**: The filament must not be in the hotend!!

Additionally, direct link to the Marlin command description. https://marlinfw.org/docs/gcode/M302.html

teachingtechYT, 2021-07-13:
The page doesn't mention M302, so I assume you are suggesting that it should?
I am strongly against this, as most people will run the test without disassembling anything, which needs the hot end up to temp.
Remember that this is aiming at a wide range of skill levels, so I need to keep everything as simple as possible.