teachingtechYT/teachingtechYT.github.io #251: Baseline print is a hard first print ever

kpishere, 2021-02-06:
So, I'm a nube who is enjoying doing things the hard way.  Started with RePap Itopie, modified SCAD (learned it) for a simpler build list of parts.  Ordered all parts directly, built it.  Modified firmware for SMART RAMPS 1.4.6 with MEGA 2560 (bricked an DUE in process ... not supposed to happen but did when playing with serial port mappings in Marlin).

Anyhow, a nube but still hands on.  In commissioning this new printer .. I've given up on this baseline calibration model as being my first print .. it is a hard one!  The ends keep curling.  There is no brim.  Spent lots of time with different bed configurations and tweaking Z-axis (a good learning experience).  Tried cold aluminum bed with U-stick, aluminum with Kapton, glass bed, and heated variants.  All had lifted corners and part eventually comes off.

The smallness and corners without brim or otherwise seems to make it challenging.

Perhaps a different calibration print that is more tolerant of a poorly configured printer?

teachingtechYT, 2021-07-13:
I would say your printer is the exception rather than the rule. This is the first report of this from the many thousands who have used the site, which suggests that this is not a difficult print. You seem to have very major bed adhesion issues if this print is causing you trouble.

closed 2021-07-13 by teachingtechYT

kpishere, 2021-07-13:
Hah!  Yes, I don't recall the specifics now but it has stabilized.  I reliably print ABS now.  I think I may have been trying to print that without an enclosure and the with the wrong temperatures.  Also, at the time, my parts were printed in PLA.  I'd replaced them with ABS parts a while back now but the PLA ones were misshapen and loose.  It does take a while to work through all of the variables when building a first printer from scratch and never having a functioning one for reference.