teachingtechYT/teachingtechYT.github.io #258: New Feature Request - outer wall speed test tower

midicdj1000, 2021-02-13:
I notice while calibrating my anycubin chiron it has noise in the outer walls and I can tweak based on the print speeds. It would be very useful to generate a tower with changing print speeds so you can fine tune for quality.
[image]

one example of changing the print speed

Gitdad, 2021-02-20:
I second this. It would be great to get a speed option added to the acceleration / jerk test. 

p.s. great work on the site!

teachingtechYT, 2021-07-13:
I will keep this in mind for a future test. I don't think it would be too hard to implement.

teachingtechYT, 2021-11-03:
I designed the CAD for this yesterday and implemented an early, manual version to torture two extruders with flexible filament.
Here is the geometry:
[image]
The aim to provide a range of shapes, while still being quick to print.
The user will set their slowest feedrate the lowest segment, then the target feedrate for the next four segments. The STL will be sliced in vase mode and use simple M220 commands to set feedrate percentage. This is the reason for vase mode, to avoid retractions also being sped up with M220.
The test worked well in my testing thus far. The single outer wall shows extrusion quality nicely on the largest curved side, and the fine details are good for exposing a lack of flow rate when things speed up.
Feedback welcome.

teachingtechYT, 2021-11-12:
Implemented!

closed 2021-11-12 by teachingtechYT

midicdj1000, 2021-11-12:
I just ran this on my chiron and it is exactly what I wanted.  Thank you so much