teachingtechYT/teachingtechYT.github.io #259: New Feature Request - Nozzle size selection for Retraction Test

casas-7, 2021-02-16:
Hello,

Would it be possible to add an option to do the retraction test using a 0.6mm nozzle?
Or is this doable using a start gcode? If so, what would be the command to do so?

Thanks a bunch mate.
Cheers!

ragd011, 2021-02-25:
Hi, I'm also wondering about different nozzle sizes.  I started running this before I thought about the fact that my nozzle is large and on a volcano.  It seems to be doing well but that doesn't give me a meaningful result since I don't know what this code is doing.  Thanks.

bringbackgroupb, 2021-02-28:
In my very brief review of the code, it looks like this site works by finding certain parts of the code and changing them, rather than generating new g-code itself. I think the easiest way to implement it would be to create base g-code files for at least 0.4, 0.6, and 0.8mm nozzles (as those are the most common). Then the site could switch which base code file to use based on the nozzle selection. Making the selection infinitely variable would basically require writing a slicer from scratch which seems well outside the scope of this site. The easiest way to do this would be for TT to generate those base files himself since then he'd have consistent settings between all the base files. I suppose someone else could do it by reverse engineering the base g-code, but who has time for that :)

Ansari-Tech, 2021-03-14:
Duplicate of #195

teachingtechYT, 2021-07-13:
Duplicate but it's on the way this week.

closed 2021-07-13 by teachingtechYT