teachingtechYT/teachingtechYT.github.io #261: Slicer Flow Calibration: Measure inside the cube

gloomytree, 2021-02-17:
It might be a good idea to tear off a few layers from the top and bottom of the cube to measure inside the cube instead of the first / last layers. The lower layer suffer from elephant foot and I seem to have bigger top layers too, for some reason. 

Measuring from the top: 0.5
Masuring inside: 0.43
First layer with elephant foot: 0.6

HermitTurtle, 2021-04-06:
I noticed the same issue. There is a significant difference between the middle and top/bottom layers. The bottom is expected; the well-known elephant's foot defect. The top was a bit of a surprise but there's easily a 0,05 to 0,10 mm difference.

Tuning for 0,4 mm; I found my middle section came very close to 0,3 mm.

teachingtechYT, 2021-07-13:
Thanks I will add a note.

closed 2021-07-13 by teachingtechYT