teachingtechYT/teachingtechYT.github.io #270: gcode prints but nothing else

JonSnow88, 2021-03-01:
Using an ender 3 v2, manually bed leveled using 4 knobs and then mesh bed level from firmware. Used the custom gcode generator to print the first layer squares and got all squares to print evenly and clean.

However when I slice anything using Prusa Slicer everything prints in the air and nothing sticks. 

I checked the slicer settings I have and they are basically the same and have added M420 S1 to my start gcode. 

The calibration cube printed with no problems as well, but when i tried the same from thingivers and prusa slicer it was the same, printing in the air. 

What am i missing?

Eliminateur, 2021-04-30:
i have this issue as well, and viceversa, if you doa nromal print and then one of the files here, it prints air

teachingtechYT, 2021-07-13:
There is nothing here that affects the first layer beyond what ABL method you select from the drop down.
In the start gcode, you should be able to see the finished sequence.
Perhaps in your slicer there is a global offset applied?
I would suggest generating a cube from here and slicing your own then comparing the gcode in a text editor. Not only the start gcode, but the initial Z values for G1 moves.

closed 2021-07-13 by teachingtechYT