teachingtechYT/teachingtechYT.github.io #272: Extruder E-steps Calibration on Malyan (Monoprice Select Mini V2)

BurtHarris, 2021-03-01:
Just a note: using the Extruder E-steps Calibration page and running that with the current official firmware (V41), the bare M92 command does not produce any output.

Addressing an over-extrusion issue was my main point in running through this calibration. I'm wondering if an alternative exists...

BurtHarris, 2021-03-01:
I found specific instructions at https://mpselectmini.com/extruder_calibration

BurtHarris, 2021-03-01:
Output from M503:

> echo:Steps per unit:
> echo:  M92 X93.00 Y93.00 Z1097.50 E97.00
> echo:Maximum feedrates (mm/s):
> echo:  M203 X150.00 Y150.00 Z1.50 E50.00
> echo:Maximum Acceleration (mm/s2):
> echo:  M201 X800 Y800 Z20 E10000
> 

> echo:Accelerations: P=printing, R=retract and T=travel
> echo:  M204 P3000.00 R3000.00 T3000.00
> echo:Advanced variables:
> S=Min feedrate (mm/s),
> T=Min travel feedrate (mm/s),
> B=minimum segment time (ms),
> X=maximum XY jerk (mm/s),
> Z=maximum Z jerk (mm/s),
> E=maximum E jerk (mm/s)
> echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
> echo:Home offset (mm):
> echo:  M206 X0.00 Y0.00 Z0.00
> echo:Invert axis: M562 XYZE
> XYZABCD+--+-+-
> echo:PID settings:
> echo:  M301 P10.59 I0.09 D4.17 C100.00 L20
> echo:  M304 P90.90 I1.28 D21.45
> echo:Filament settings: Disabled
> echo:  M200 D1.75
> echo:  M200 D0
> ok N9040 P15 B15
> 

It seems simple enough to use the value on the 2nd line...

BurtHarris, 2021-03-01:
My measurement to the mark was 32 mm, which gave a calculated adjusted value of E110.23.   Does increasing the M92 E value increase or decrease extrusion?

teachingtechYT, 2021-07-13:
Increasing the e steps value also increases the actual filament delivered.
Good problem solving. This site works for 90% plus of printers but there's always going to be anomalies.

closed 2021-07-13 by teachingtechYT