teachingtechYT/teachingtechYT.github.io #280: Temperature Tower Mirroring issue - Leads to failures

rickalm, 2021-03-23:
When the GCode for the tower is created the order of the operations is not a mirror. As a result the extruder head lingers on the right tower overhang for a bit, then moves Z-up preping for the next layer which causes the overhang on the right side to become messy and that kinda invalidates the whole bridge point of the test. The left side origin and exit is different (e.g. not a mirror)

[image]

rickalm, 2021-03-23:
This is another print from about 2 weeks ago with different filament and you can see the same effect. I use the "5 Squares" to confirm my bed alignment

[image]

rickalm, 2021-03-23:
Bowden Extruder, Used my slicer settings on generator .. 6mm retract 30mm/sec

ivan5rov, 2021-06-11:
I am having the same problem but with direct drive (0.5mm retract at 40mm/s). Thought it might be cooling issue, but adding external part fan did not make any difference. I noticed this test introduces a Z-hop (is it intentional?) Could be due to speed, as test runs fairly fast. I wonder if possible to have speed control on the generator page to select speed as well

teachingtechYT, 2021-07-13:
The site is undergoing a big update that will be released later this week. This test will have he ability for the user to select z hop or leave it at zero. Hopefully this fixes your problem.

closed 2021-07-13 by teachingtechYT