teachingtechYT/teachingtechYT.github.io #283: Followed setup bltouch probe still not deploying/turning on

Wnwhite, 2021-03-28:
I recently moved and had dissemble my printer which I thought was a good time to upgrade the motherboard and the display. I am now working with a btt skr 1.4turbo and a tft35 e3 v3.0 with a BLtouch v3.1. I am able to flash the firmware on both the mobo and the tft but the BLtouch does not activate. I am able to control the board and home the x/y axes, heat the bed and hot end, as well as send commands to the mobo through the tft terminal. I tried both the methods described on the techingtech's page using the recommended wiring and marlin changes with now two different BLtouch's. upon start up the tft read "no printer attached" for about 20 sec then I get 2 beeps and an error message "unable to enable bed leveling". Any suggestions would be very appreciated.

Wnwhite, 2021-03-29:
I was able to successfully use the manual mesh bed leveling and the printer is currently working. So I thought it must be some problem with the firmware. The BLtouch V3.1 needs 5 V logic so I uncommented the 5V logic "BLTOUCH_SET_5V_MODE" this also resulted in the Bltouch not deploying and the LEDs remained off.

chiral-lab, 2021-03-30:
I have a BTT SKR Mini E3 V2.0 and a TFT35 E3 3.0 and the "no printer attached" message is normal behavior. The display seems to show it when the mainboard is still booting. When I installed my BL-Touch the LEDs have been working once the wiring was correct, even when the firmware still had BL-Touch completely disabled. Perhaps you should doublecheck your wiring again and compare it with the Board schematic?

Wnwhite, 2021-04-01:
So I gave it another try today. looking at the pin diagram today the 3rd unused pin in the z endstop port is labeled 5V plugging the white wire in that port resulted in the red led flashing! which was exciting but that meant there was a hardware/wiring issue and again my tft wouldn't turn on. Plugging the white wire back into the 1.27 pin results a solid red and blue LED but the probe deploys and retracts seeming correctly. (the wiring which is now "working"  was the wiring I had been trying which was shown in the video and the guide before with no success) The combination red/blue LED means "Wiring defect indicator : Blue and Red LED for indicating wiring defects." the amazon seller i purchased the Bltouch from suggested switching the yellow and red wires which makes the whole thing not work. I enabled the BLTOUCH_FORCE_MODE_SET and switching the probe to 5V causes the blue light to go off but switching back to the touchscreen causes the blue light to turn back on. I then went back to the marlin menu and set the 5V mode and saved it to EEPROM and tried going back to the touchscreen. Again the blue light came on, so I set the voltage to 5V using BLTOUCH_SET_5V_MODE and commented out BLTOUCH_FORCE_MODE_SET and reflashed the firmware and the red/blue light is still on. I tried setting the p offset using the z offset wizard function and I get a bltouch error and says I need send a M999 command to reset my printer where the printer homes x/y then reads error before the z-axis moves. This also happens when trying to run the auto bed leveling function. At this point I am betting on either I was shipped a defective bltouch or I killed it.

teachingtechYT, 2021-07-13:
I wish you well but please don't post issues unrelated to my website here.

closed 2021-07-13 by teachingtechYT