teachingtechYT/teachingtechYT.github.io #296: First layer: additional info for V rollers and ABL

pogof, 2021-04-27:
Im sure its mentioned somewhere that you should adjust the eccentric nuts on the Z ("X"?) axis. However if adjusted poorly (too tight or too loose you can get uneven results with ABL as one side of the axis is slightly behind the other one (on ender 3 for example where there is only one leadscrew). It was too tight and with my digital level measured only 0.1° difference, yet you could actually see it. After loosening the nut a bit all 5 squares look the same.

teachingtechYT, 2021-07-13:
I agree with you. V rollers are mentioned on the frame check tab, which is mean to be followed before anything else.
[image]

teachingtechYT, 2021-07-13:
I've added an extra sentence to that section to reflect your experiences.

closed 2021-07-13 by teachingtechYT