teachingtechYT/teachingtechYT.github.io #310: Priming the nozzle in the First Layer calibration

atsushi1, 2021-06-04:
It took me many iterations and many mis-adjustments before realizing the skirt of the first square was grossly underextruded because my nozzle was dripping during the warming up process and not because of the bed leveling. Can you add a coment about that in the instructions and/or add some nozzle priming code at the beginning. 

I looked at the G-code and there also seems to be some custom retraction code, "G1 E-5 F2400 ; custom retraction" which may not be un-retracted by the priming code for some reason: 
G1 E-5 F2400 ; custom retraction
; skirt 1
G1 Z0.400 F1200
G1 X24.636000000000003 Y26.314 F6000
G1 Z0.200 F1200
G1 E0 F2400 ; custom un-retraction/prime
G92 E0.0000

Before you dig too much into this, I should say this may be an issue with my brand new Creality Ender 3v2 running fw version 1.0.2. Can you verify if you have this issue with other printers?

atsushi1, 2021-06-05:
I do want to emphasize that these test programs are an amazing resource. I really appreciate you putting them together for the community!

I just ran it on my Velleman K8200 and same behavior. No priming of nozzle and very thin skirt on the first square closest to home... But other than the skirt, it does tell me my bed is pretty level... (No ABL on either printer).

teachingtechYT, 2021-07-13:
Thanks for the feedback. Everything you want is already possible and explained on each generator form.
[image]
The lack of the unretract for the first move should be fixed in an update out later this week.

closed 2021-07-13 by teachingtechYT