teachingtechYT/teachingtechYT.github.io #318: New approach to filament drying

jon-bondy, 2021-07-06:
I looked for a way to create a proposal or suggestion, but could not find it, so I felt compelled to create an issue.

This appeared in an OpenSCAD discussion forum recently:

wrt filament drying, it seems a bit daft,  to keep a whole spool of filament dry, when all you need is the few metres you are printing. It is OK, if you are, say, continuously printing the same filament type, but when experimenting with different colours/types, you can end up with a dozen or so spools to keep dry. There are plenty of solutions, vacuum sealing/desiccant, etc. but it is a bit of a hassle/expense. Why not keep the spools as is, no packaging, and condition what you need as you use it?  So, we could use a relatively small diameter, slightly heated tube, heating element wrapped around the outside, holes in tube to let the moist warm air escape - a foot or so ptfe perforated bowden tube would most likely do, or possibly a larger tube with warm air blown through.

I wonder if research into this would be interesting and helpful.

Jon

teachingtechYT, 2021-07-13:
I'm all for suggestions but this is not the place. Please use the contact page on this site and select request or misc.
impossible for to keep track of any site unrelated queries here.

closed 2021-07-13 by teachingtechYT

jon-bondy, 2021-07-13:
Sorry to have wasted your time.

If I could have found a Contact page on this site, I would have used it.  Even having been directed there, I cannot find it.

teachingtechYT, 2021-07-13:
No time wasted. Please understand due to the volume of contact I receive through YouTube comments, emails, here, Patreon, Thingiverse, etc it is impossible for me to work efficiently unless I had some structures in place.
[image]
https://teachingtechyt.github.io/contact.html