teachingtechYT/teachingtechYT.github.io #322: Visual studio code error when compiling marlin

Joost79, 2021-07-11:
Not sure if this is the right place, forgive me I'm a noob...

I've watched your video and read the tutorial about compiling marlin 2.0 firmware for my ender 3d pro (on my 8bit v114 board). I've installed the visual studio, downloaded the marlin bugfix firmware and the correct example config.

Copied it to the right directory and loaded the directory in visual studio code on my windows 10 pc. So far no issues, and if i would just compile the firmware from there, everything works. But after i updated the original firmware to the settings as suggested in this page (https://teachingtechyt.github.io/upgrades.html#bltouch) and compile again, it fails every time, without exceptions.

I have quadruple checked everything and as far as i can find i have correctly removed all "//" and nothing more. Below is a screenshot of the issue;

[image]

Any input would be greatly(!) appreciated, as seem unable to solve this issue.

teachingtechYT, 2021-07-12:
This is completely unrelated to this site.

closed 2021-07-12 by teachingtechYT

Joost79, 2021-07-13:
My apologies, as i stated in my first sentence, i’m new to this.

So where should i go with this? My searches didn’t result in any answers.

Thanks

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10