teachingtechYT/teachingtechYT.github.io #323: README before submitting issues

teachingtechYT, 2021-07-15:
Please read and adhere to the following guidelines when posting a new issue:

1. Please use the search feature before posting, including already closed issues. Some features have been requested and turned down already with an explanation. Duplicate issues will be closed.
2. Use a descriptive title. If posting a feature request, include [feature request]. Same goes for bugs, typos, etc. Vague titles blend in and are less likely to be addressed.
3. Do not use this issue section to seek technical support unrelated to the site. Such issues will be closed immediately.
4. Please understand that although I'm all about improving the site and adding functionality to help as many people as possible, a balance needs to be found to ensure ease of use for the average user. If a new feature will marginally improve the experience for 0.5% of users, but confuse 40% of others, the feature is best omitted.

Thanks for reading before posting and for the significant community contributions that have improved this site.

XxxMikeyxxX, 2022-02-23:
Spelling correction needed: Under Slice flow calibration. Almost at the bottom:  For example, using M221 S90 would tell the firmware to only extrude 90% of what the gcode asks for. This is an easy method for making a quick correction that will alow the tests to complete successfully.

alow should be "allow"

zerquetch, 2024-03-12:
How does this work with Klipper? The Fluidd console doesn't recognize gcode commands: / {"code":"key61, "msg":"Unknown command:M92", "values": ["M92"]}

zerquetch, 2024-03-13:
I've searched. I've read. I've watched. Either I'm searching with the wrong criteria or I'm flat out stupid. Either of these being the case, I don't ask for help very often. In 4 years of 3d printing, I've never directly asked anyone a question, other than rhetorically or with sarcastic humour. The fact remains, I don't know how to proceed. If you feel I'm wasting your time, I apologize. Thank you for all the knowledge you pass on, I am, and will remain, a student of this craft.