teachingtechYT/teachingtechYT.github.io #327: Organisation of tabs for more intuitive workflows

Pun-e, 2021-07-16:
I love your site, and it is the # 1 thing I give to all new printers in how to best go from noob to pro.

In that vein, I've always been puzzled by the layout of the tabs, and I'm sure it's tripped up many young players.

[image]

The first few are fine, but I feel like Acceleration, XYZ steps cal., and motor current should be towards the first-time and only do once tests like the frame check and PID autotune. Similarly the filament tuning tests (being used much more frequently) should be grouped together and in a more logical order according to their dependencies. I.e. Temperature > Linear Advance > Retraction

Here's my _perfectly splendid_ mockup:

[image]

As an advanced user, I would appreciate the ability to hide all of the tabs I'm not using, and I hope that a new user would appreciate being able to see what they should do at a quick glance.

My only sore point with my mock-up are the "diagnostic prints" as my sleep-deprived brain couldn't find a better place for them than after the intro. But hopefully that's enough to communicate my idea and get your gears whirring.

Thanks again for this great resource, it's truly a game-changer for the new 3d-printer (the device and the user)

teachingtechYT, 2021-07-19:
I like the idea of categorization but everyone uses the site in a different way and that makes it hard to agree.
For instance, i have PID autotune at the start because because I see it as a safety essential to tick off.
I also approach the site from a point of view of changing mainboard/firmware and therefore some of the steps are dictated by this process.
Hiding the tabs in a hard no I'm afraid. No doubt someone will hide some by accident and have no idea where they went.

Pun-e, 2021-07-22:
I think my use of the word "hide" was misguided, I meant the ability to collapse a category so only the heading is visible, and the contained tabs are temporarily hidden from immediate view when its parent category is collapsed - think how most touch interfaces look these days.

But either way I respect your opinion and understand the minefield that is UI design

closed 2021-07-22 by Pun-e

teachingtechYT, 2021-07-22:
Although you've closed this, things are in motion and I do plan to introduce some categorisation, just a little different to this. THanks for your input, it is valuable.