teachingtechYT/teachingtechYT.github.io #339: Temperature timeout after long ABL routine [Minor]

tycho94, 2021-07-29:
Hi,

I have quite a slow ABL routine and noticed that my set temperatures timeout (I'm fairly certain they are default marlin timeouts) near the end of the ABL routine. It could also be possible that this is not a timeout but an issue with my printer not having stable temps during ABL (during prints temp stability is perfectly fine).

How I currently solve this is after the ABL routine I run the m140/104 and m190/109 once again to solve this. Perhaps this could be an improvement to the generated gcodes. In the worst case, it might solve some issues others could also be expierencing. And in any other case it will not influence the print as the temps are already at the right values.  

E.g. 

`M140 S60 ; custom bed temp
M190 S60 ; custom bed temp
M104 S200 T0 ; custom hot end temp
M109 S200 T0 ; custom hot end temp
G28 ; home all axes
G29 ; probe ABL
M140 S60 ; custom bed temp
M190 S60 ; custom bed temp
M104 S200 T0 ; custom hot end temp
M109 S200 T0 ; custom hot end temp`

teachingtechYT, 2021-07-29:
I'm currently working on a tweak that lets the user strip all of the default start gcode and use only what they enter into the custom box. I think this should be able to accommodate your situation.

teachingtechYT, 2021-07-30:
The above update is now live.

closed 2021-07-30 by teachingtechYT